openmv/openmv-arduino-rpc #2: Transfer images from the Arduino Portenta over Wifi

TannerGilbert, 2020-12-26:
I want to transfer images from my Arduino Portenta H7 to my computer over Wi-Fi using OpenMV, and I found the `image_transfer_jpg_as_the_remote_device_for_your_computer.py` example, but I get the following errors when running the script.

```
Could not open "/media/gilbert/2821-0317/rpc.py" for reading. Either the file does not exist or you do not have the permissions to open it
```

and

```
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "rpc.py", line 560, in <module>
  File "rpc.py", line 590, in rpc_wifi_master
AttributeError: 'module' object has no attribute 'WINC'
```

As I want to use WIFI, I changed the interface to WIFI as described in the code. Any help is much appreciated.

kwagyeman, 2021-05-29:
Never saw this until now. Yes, I have to update RPC to include the Portenta WiFi module.

sw4p, 2021-08-07:
I am facing the same issue.

iabdalkader, 2021-08-07:
The WiFi module on Portenta is called WLAN, also the API is slightly different the WINC's API, see the WiFi examples for Portenta. I can make the WINC module compatible with the Portenta WiFi module (which is also the module used by official mpy boards) so the same scripts work with both modules (please open an issue). That said, this module will still need to be updated.

sw4p, 2021-08-07:
Opened an issue - https://github.com/openmv/openmv/issues/1427

Thank you!

kwagyeman, 2021-12-12:
This was fixed.

closed 2021-12-12 by kwagyeman