openmv/openmv-arduino-rpc #4: CAN.h include issue for ESP32

chacalnoir, 2021-02-06:
Hello, great library. Thank you for developing and providing it!
I ran into an issue using this library with an Adafruit HUZZAH32 - ESP32 Feather (https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview). When compiling, CAN.h could not be found. I made the following modifications to your library to get the software to compile and run:

openmvrcp.h
11: #if !defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ARCH_ESP32) && !defined(ARDUINO_ARCH_NRF52840)
220: #if !defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ARCH_ESP32) && !defined(ARDUINO_ARCH_NRF52840)
openmvrcp.cpp
680: #if !defined(ARDUINO_ARCH_ESP8266) && !defined(ARDUINO_ARCH_ESP32) && !defined(ARDUINO_ARCH_NRF52840)

With those changes, the software compiles and runs correctly. I do not know whether this is an issue for all ESP32s since I have only tested with the Adafruit feather. Further, there are CAN libraries for ESP32 (such as https://github.com/miwagner/ESP32-Arduino-CAN) that provide CAN.h, so that could be used as a dependency instead of removing the CAN support in your library like I did.
Either way, I just wanted to alert you to this since you likely have not had a chance to test with an ESP23 Feather yet.

Thank you!

mengguang, 2021-08-21:
Same problem on STM32 Arduino.  
May be It's better to add a switch macro like ENABLE_CAN:  
```
#ifdef ENABLE_CAN
#include <CAN.h>
#endif
```
If this is acceptable, I can make a PR.

kwagyeman, 2021-12-12:
Hi, this has been fixed.

closed 2021-12-12 by kwagyeman