openmv/openmv-arduino-rpc #7: Issue using I2C on the Portenta H7

BenjaminDannegard, 2021-11-26:
I am trying to use I2C with the rpc library on the Portenta H7. But I get the error: 

File "<stdin>", line 3, in <module>
File "rpc.py", line 365, in __init__
ValueError: I2C(2) doesn't exist

I get this error when I try to upload the "popular_features_as_the_controller_device.py" example script to the Portenta H7.

kwagyeman, 2021-12-02:
Hi, the library is hardcoded just for the OpenMV Cams. This line needs to be changed here and the firmware re-built: https://github.com/openmv/openmv/blob/master/scripts/libraries/rpc.py#L395

Making the i2c port to be used a constant would be the fix.

BenjaminDannegard, 2021-12-07:
Would it be possible to update the library so that it works without changing that line in the library?

kwagyeman, 2021-12-12:
https://github.com/openmv/openmv/pull/1473

closed 2021-12-12 by kwagyeman