openmv/openmv-arduino-rpc #8: Issue with CAN and UART functions for the "openmvrpc" library in Arduino IDE

BenjaminDannegard, 2021-12-01:
I am trying to upload the example sketch: "arduino_to_arduino_communication_as_the_remote_device.ino" from the library to a Portenta H7 and I get the issue:

In file included from C:\Users\benja\Documents\Arduino\RPCtest\RPCtest.ino:2:0:
C:\Users\benja\Documents\Arduino\libraries\OpenMV_Arduino_RPC\src/openmvrpc.h: In member function 'virtual void openmv::rpc_can_master::begin()':
C:\Users\benja\Documents\Arduino\libraries\OpenMV_Arduino_RPC\src/openmvrpc.h:229:37: error: 'CAN' was not declared in this scope
     virtual void begin() override { CAN.begin(__bit_rate); CAN.filter(__message_id); }
                                     ^~~
C:\Users\benja\Documents\Arduino\libraries\OpenMV_Arduino_RPC\src/openmvrpc.h:229:37: note: suggested alternatives:
In file included from C:\Users\benja\Documents\Arduino\libraries\OpenMV_Arduino_RPC\src/openmvrpc.h:12:0,
                 from C:\Users\benja\Documents\Arduino\RPCtest\RPCtest.ino:2:
C:\Users\benja\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\2.6.1\cores\arduino/mbed/drivers/include/drivers/CAN.h:39:7: note:   'mbed::CAN'
 class CAN
       ^~~

Adding: #define ARDUINO_ARCH_NRF52840 to the openmvrpc.h file fixes the issue for the Portenta H7, allowing for the sketch to be uploaded.

My main purpose is to upload the sketch to a Nicla Sense ME, and that fix does not help when trying to upload to the Nicla. It still gets errors for the UART functions in the library. Could the library be updated to work with the Nicla Mbed core?

kwagyeman, 2021-12-02:
Hi, the library has never been tested on the Portenta. Is it possible you could send a PR using their CAN library?

BenjaminDannegard, 2021-12-07:
Hey, I am not sure what you mean with PR. Could you provide instructions so I can send you that?

kwagyeman, 2021-12-12:
Fixed

closed 2021-12-12 by kwagyeman